NicoHood/MSGEQ7 #5: Use of this lib with Esp8266

marksev1, 2016-04-15:
Hi,.. I'd like to use your lib on a esp8266. Since esp has just one adc I would use just one msgeq7. Can I still input stereo signal and sample it as mono? Or just maybe buy a stereo to mono adapter? Also since there is just one ADC i would just look at one frequency for example some lower bass ones. Could you help me how to use your library for esp8266 pretty please.

NicoHood, 2016-04-15:
ESP has a 1.8V ADC as far as I remember. Might be different. You can use a single MSGEQ7 at maximum, but you'd need to shift down the voltage. I have not idea if it works, but let me know of further information.

marksev1, 2016-04-15:
What about from the coding perspecive could you guide me a bit on how to use it.

NicoHood, 2016-04-15:
Just use it in mono mode and see if it compiles? The led example should work I think? any compile errors? Have you even tried it?

NicoHood, 2016-04-16:
I just tried to compile the sketch and it seems to work. You just need a voltage divider and try it out. If it really does not work, come back and comment. And if it does, also come back and comment so we all know the result :) Thanks and good luck!

closed 2016-04-16 by NicoHood

marksev1, 2016-04-17:
Cool I will do so. :) With the example sketch. Next if I see that it works I have some sketch with neopixelbus library for animations and mqtt support is also inside, and I would like to put your library inside. So I'll gonna probably ask you a few questions then, since I'm a beginner.

NicoHood, 2016-04-17:
Sure, but please do it via tox or PGP mail, so this issue tracker is not spammed.
www.contact.nicohood.de