NicoHood/MSGEQ7 #6: MSGEQ7.get(3) noise

lh84, 2016-06-18:
Hi Nico,

thanks for the great lib. Works well, besides one issue.
I read the MSGEQ values in a for loop. And when the programm starts and music is playing everything is fine. But after I stop the music frequency band 3 (MSGEQ7.get(3)) gets always high noise. 

Following my sketch:

`
/*
  Copyright (c) 2014-2016 NicoHood
  See the readme for credit to other people.
  MSGEQ7 Led example
  Output via Led (choose a PWM pin!)
  Fast setup to demonstrate functionality
  Reads MSGEQ7 IC with 7 different frequencies from range 0-255
  63Hz, 160Hz, 400Hz, 1kHz, 2.5kHz, 6.25KHz, 16kHz
*/

// MSGEQ7
#include "MSGEQ7.h"

#define pinAnalog A1
#define pinReset 7
#define pinStrobe 12
#define MSGEQ7_INTERVAL ReadsPerSecond(50)
#define MSGEQ7_SMOOTH 150// Range: 0-255

CMSGEQ7<MSGEQ7_SMOOTH, pinReset, pinStrobe, pinAnalog> MSGEQ7;

#define DEBUG

#ifdef DEBUG
 #define DEBUG_PRINTLN(x)  Serial.println(x)
 #define DEBUG_PRINT(x)  Serial.print(x)
#else
 #define DEBUG_PRINT(x)
 #define DEBUG_PRINTLN(x)
#endif

uint8_t level[7]; 
const int LEDpins[7] = {3,3,5,6,9,9,10};

void setup() {
  Serial.begin(115200); 
  // This will set the IC ready for reading
  MSGEQ7.begin();
  // led setup
  for (int i = 0; i < 7; i++) {
    pinMode(LEDpins[i], OUTPUT);
  }
}

void loop() {
  // Analyze without delay every interval
  bool newReading = MSGEQ7.read(MSGEQ7_INTERVAL);
  // Led output
  if (newReading) {
    // Read bass frequencies   
    for (int i = 0; i < 7; i++) {
        level[i] = MSGEQ7.get(i);
        if(i==3) {
          //remove noise from mid frequency
          level[i] = mapNoise(level[i], 40, 250);
        } else {
          level[i] = mapNoise(level[i]);
        }
        // Output PWM signal via Led to the music beat
        analogWrite(LEDpins[i], level[i]);
        DEBUG_PRINT(mapNoise(level[i]));
        DEBUG_PRINT("   ");
    }
    DEBUG_PRINTLN("");
  }
}
`

sample Serial output while music playing: 
96   250   251   255   67   75   0   
79   251   252   255   142   149   68   
78   252   255   255   188   193   125   
90   255   255   255   215   200   142   
106   239   252   255   218   184   136   
103   246   255   255   205   160   120   
98   249   251   255   217   166   99  

sample Serial output while music is NOT playing: 
0   0   0   244   0   0   0   
0   0   0   245   0   0   0   
0   0   0   245   0   0   0   
0   0   0   244   0   0   0   
0   0   0   245   0   0   0   
0   0   0   244   0   0   0   
0   0   0   245   0   0   0   
0   0   0   243   0   0   0  

Do you know why the noise is that high for band 4?

NicoHood, 2016-06-18:
What happens if you unplug the cable? To you then also get those high noises? Try to disable smoothing (0). Maybe there is something wrong in the function. I have to check this myself.

Also you use a different mapNoise() range. Maybe there is the bug. Can you remove the if for i=3 and test again?

lh84, 2016-06-19:
I modified the loop as follows:

```Javascript
#define MSGEQ7_SMOOTH 0
...

void loop() {
  // Analyze without delay every interval
  bool newReading = MSGEQ7.read(MSGEQ7_INTERVAL);
  // Led output
  if (newReading) {
    // Read bass frequencies   
    for (int i = 0; i < 7; i++) {
        level[i] = MSGEQ7.get(i);
        // Output PWM signal via Led to the music beat
        analogWrite(LEDpins[i], level[i]);
        DEBUG_PRINT(mapNoise(level[i]));
        DEBUG_PRINT("     ");
    }
    DEBUG_PRINTLN("");
  }
}
```
## Samples - with Serial.print(mapNoise(level[i]))
With unplugged cable:
```
46     9     9     56     2     2     0     
44     9     6     57     1     1     0     
43     11     10     56     2     0     0     
45     9     6     56     0     2     0     
```
With plugged cable and music playing:
```
126     52     77     62     83     51     7     
97     39     111     61     65     45     18     
81     31     90     58     64     35     13     
60     21     79     61     71     30     7     
```
With plugged cable and music is not playing:
```
0     0     5     57     6     9     4     
0     0     4     58     4     7     2     
0     0     5     58     5     6     4     
0     0     5     58     4     8     1     
```

## Samples - with just Serial.print(level[i])
With unplugged cable:
```
78     33     32     88     28     24     20     
78     33     28     87     25     23     18     
77     30     34     87     32     26     21     
76     31     30     87     28     23     20     
```
With plugged cable and music playing:
```
83     46     44     89     102     93     35     
74     40     41     87     84     77     30     
82     58     63     88     111     123     37     
90     56     91     89     112     108     35      
```
With plugged cable and music is not playing:
```
13     17     23     82     23     21     21     
10     16     24     83     24     24     18     
13     16     24     83     22     24     18     
13     16     24     82     22     23     17     
```

As you see there is a lot of noise in the mid frequencies. Thats why I used the mapnoise() function.

NicoHood, 2016-06-19:
And what if you completely unplug the cable?

NicoHood, 2016-06-19:
Going to check that myself....

NicoHood, 2016-06-19:
I used this sketch:

```
[...]

void setup() {
  Serial.begin(115200);
  MSGEQ7.begin();
}

void loop() {
  // Analyze without delay every interval
  bool newReading = MSGEQ7.read(MSGEQ7_INTERVAL);

  if (newReading) {
    for (int i = 0; i < 7; i++) {
      Serial.print(MSGEQ7.get(i));
      Serial.print('\t');
    }
    Serial.println();
  }
}
```

And I get the following outputs (cable plugged in, no music playing, also with cable plugged out):
```
13	15	13	13	15	15	16	
14	16	13	14	15	15	17	
15	15	12	15	14	14	16	
13	16	13	14	15	15	17	
14	15	12	13	14	14	17	
14	15	12	14	15	15	17	
14	16	12	14	15	14	17	
13	16	11	14	15	14	17
```

With mapnoise all levels are at zero.
Can you please try this sketch?

There are a few possible reasons for this error:
* An MSGEQ7 IC is broken (I had one and it took me forever to find the reason)
* Wrong hardware setup
* Ground noise from the PC (since it also happens with not cable attached)

I suggest to:
write a sketch that checks if channel 3 > 50 and light an led (aka error). Then I'd plug the arduino with no music cable attacked into an usb PSU (but not the pc). This way you can check if its a ground noise problem. I'd try differen PSUs and different pcs. Only connect an usb cable to the arduino. If it still exists there is a wrong setup (wrong cacapcitor or cables) or a defective msgeq7.

You can also try one of my boards. I can sent you all components (unsoldered!) with pcb for a few bucks if you want to.

lh84, 2016-06-19:
Hi Nico,

Thanks for your help. All the components are right in place. I ordered 5 msgeq from China. I guess the msgeq is over.

How much is your "bausatz" :)?

Cheers from Berlin

NicoHood, 2016-06-19:
So another msgeq7 did the trick?

10€ inclusive (world wide) shipping for a pcb + all components without soldering. I got a few left, like 5 or so.

closed 2017-07-05 by NicoHood