NicoHood/MSGEQ7 #7: Suggested change to text in documentation

gptix, 2017-06-16:
I suggest changing the text (at https://github.com/NicoHood/MSGEQ7)

from

"The MSGEQ7 is an IC designed to analyze music on different frequencies. This library reads the MSGEQ7 with 7 different frequencies from range 0-255, stereo or mono. It can be used to visualize audio through Leds for example. This repository also includes the source of an Breakout Board.

The frequencies are: 63Hz, 160Hz, 400Hz, 1kHz, 2.5kHz, 6.25KHz, 16kHz"

to

"The library in this repository provides an interface to one or two MSGEQ7 chips, to convert analog voltage output from the chips to digital data. The data is of values from 0 to 255 (8 bits) for one (mono) or two (stereo) "channels".  A channel is the output from one MSGEQ7 chip.

The MSGEQ7 is an IC designed to analyze sound by frequency. The chip measures aspects of input in seven frequency bands, and can provide output related to one (at a time) of these bands as an analog DC voltage signal. The frequency bands the chip analyzes are centered on: 63, 160, 400, 1K, 2.5K, 6.25K, and 16H Hertz.

This repository also includes source information for a breakout board with two MSGEQ7 chips."

NicoHood, 2017-06-16:
Sounds good. I just notived that mono and stereo is also wrong, as my library supports infinite msgeqy7 channels. could you also include this and open a PR for that? I'd directly merge it :)

gptix, 2017-06-17:
Will do, just now.

 

I got into this because a friend of mine is using your library and FastLED to make an EQ display built into a translucent chestplate, to wear at EDM events weith a daft-punky helmet.

 

He was getting confused about the relationship between frequency bands, channels, physical chips, and mono/stereo.

 

Take Care,

 

Jud

 

Sent: Friday, June 16, 2017 at 6:53 PM

gptix, 2017-06-17:
The library in this repository provides an interface to an arbitrary number of MSGEQ7 chips, to convert analog voltage output from the chips to digital data. The data is of values from 0 to 255 (8 bits). Output from each chip, which is input to the library, is referred to in the documentation for this library as a "channel".

The MSGEQ7 is an IC designed to analyze sound (actually, any signal of varying voltages; in the case of sound or music, a signal as from a microphone).The chip measures aspects of input in seven frequency bands, and can provide output related to one (at a time) of these bands as an analog DC voltage signal. The frequency bands the chip analyzes are centered on: 63, 160, 400, 1K, 2.5K, 6.25K, and 16K Hertz.

This repository also includes source information for a breakout board with two MSGEQ7 chips. In a case of using two channels of output from a breakout board, referring to one channel as "mono" and two channels as "stereo" makes sense.

NicoHood, 2017-07-05:
Fixed in https://github.com/NicoHood/MSGEQ7/commit/d38bde374928b0f45e6c956cdb93b69d31eeabba
Thanks for the suggestion!

closed 2017-07-05 by NicoHood