NicoHood/MSGEQ7 #8: Same value for all bands

Triscus, 2017-06-27:
In use is an Arduino Nano, an op-amp [TL071](http://www.ti.com/product/TL071) and a MSGEQ7 IC using this library. 

When I read the individual bands, all bands seem to have the same value (or nearly the same). I already double checked everything, tried with and without the op-amp and also changed the IC itself, but nothing seems to help. 

This is the code I'm using:

```
#include "MSGEQ7.h"

#define SMOOTH true
#define SOUND_IN A7
#define RESET_PIN 11
#define STROBE_PIN 10
#define MSGEQ7_INTERVAL ReadsPerSecond(50)

CMSGEQ7<SMOOTH, RESET_PIN, STROBE_PIN, SOUND_IN> MSGEQ7;

void setup() {
  // put your setup code here, to run once:
    MSGEQ7.begin();
    Serial.begin(115200);

}

void loop() {
  // put your main code here, to run repeatedly:
  // Analyze MSGEQ7 ICs now. Will always return true.
//  bool newReading = MSGEQ7.read();

  // Analyze without delay every interval
  bool newReading = MSGEQ7.read(MSGEQ7_INTERVAL);
if(newReading){

for (int i = 0; i < 7; i++)
{

  uint8_t input = MSGEQ7.get(i, 0);
  Serial.print(input);
  Serial.print(" ");
  
}
Serial.println();
}
  
}
```

The Serial output while playing the open E string:

```
36 35 36 36 35 36 36 
26 0 0 0 0 0 0 
15 14 0 0 0 0 0 
79 79 80 0 0 0 0 
81 80 81 0 0 0 0 
69 69 69 68 0 0 0 
69 68 69 69 67 0 0 
50 51 51 50 51 49 0 
83 83 83 83 83 83 83 
62 62 63 62 62 62 62 
46 46 46 46 46 46 46 
31 31 31 31 31 31 31 
50 50 49 50 50 49 50 
51 51 50 51 51 50 51 
65 65 65 65 65 65 65 
66 66 66 66 66 66 66 
53 53 53 53 53 53 53 
62 62 63 62 62 62 62 
46 46 46 46 46 46 46 
87 87 87 87 87 87 87 
65 65 65 65 65 65 66 
67 67 67 67 67 67 67 
51 51 51 52 51 51 52 
85 85 85 85 85 85 85 
86 86 86 86 86 86 86 
98 99 98 98 99 98 98 
99 100 99 99 100 99 99 
76 76 76 76 76 76 76 
85 85 85 85 85 85 85 
64 65 65 65 65 65 64 
120 119 120 120 119 120 120 
92 93 92 92 93 92 92 
80 80 80 81 80 80 80 
81 81 81 81 81 81 81 
108 107 107 108 107 107 107 
108 108 108 108 108 108 108 
84 84 84 84 84 84 84 
62 63 63 62 63 63 62 
45 46 46 46 46 46 47 
72 72 72 73 73 73 73 
55 56 55 56 57 56 56 
78 77 77 78 77 78 78 
57 58 58 57 58 58 58 
45 45 45 45 45 45 46 
48 48 48 48 48 48 48 
34 34 34 34 34 34 34 
105 106 105 105 106 105 105 
82 81 82 82 81 81 82 
61 61 61 61 61 61 61 
44 44 44 44 44 44 44 
80 80 80 80 80 80 80 
61 61 61 61 61 61 61 
54 54 54 54 54 54 54 
42 43 43 42 43 43 42 
54 54 54 54 54 54 55 
48 48 48 48 48 48 48 
33 33 33 33 33 33 33 
110 111 111 110 111 111 110 
85 85 85 85 85 85 85 
64 64 64 64 64 64 64 
46 46 46 46 46 46 46 
57 57 57 57 57 57 57 
42 42 42 42 42 42 42 
28 28 28 28 28 28 28 
28 28 28 28 28 28 28 
46 46 46 46 46 46 46 
33 33 33 33 33 33 33 
21 21 21 21 21 21 21 
99 99 99 99 99 99 99 
75 75 75 75 75 75 75 
55 55 56 55 56 56 55 
0 0 0 0 0 0 0 
55 0 0 0 0 0 0 
56 56 0 0 0 0 0 
41 41 40 0 0 0 0 
52 52 52 52 0 0 0 
37 38 37 37 37 0 0 
32 32 33 32 32 32 0 
21 21 21 21 21 21 0 
11 12 12 12 12 12 11 
5 5 5 5 5 5 5 
12 12 12 12 13 12 12 
37 37 37 37 37 37 37 
27 26 26 27 27 27 27 
15 15 15 15 15 15 15 
8 8 8 8 8 8 8 
1 1 1 1 1 1 1 
0 0 0 0 0 0 0 
10 10 10 10 10 10 10 
11 11 10 11 11 10 11 
```
Any help would be highly appreciated.

Regards 

Triscus

NicoHood, 2017-06-28:
Might be a wrong connection or a damages chip. Have you checked everything is wired correctly with the correct capacitors?

Triscus, 2017-07-03:
Yes, I have built the the ciruit multiple times with the wiring from your github readme and the wiring from the datasheet, had no luck. For the 100nF cap I am using electrolyte caps with polarity given in the data sheet. I even replaced the chip (it was from the same order). 

Here is a plot of the frequency response while sliding from 0 to 20kHz with a sine wave generator. you can see that all frequenzcies 

[image]

I am about to think that the chip is damaged, but two at the same time is bit nlikely. Any idea how I can test that the chip is working before I order a new one?

NicoHood, 2017-07-03:
The question is if there is a difference if no audio is available. So does the chip actually record anything or does it only output noise?

I dont know what you mean by caps with polarity. It should not matter where + and - is. Maybe you used wrong components?

Also I'd try different sound output from different devices at different levels. You can just test, it seems to be a hardware issue.

Triscus, 2017-07-03:
The chip reacts to the volume. If there is no input, the values are zero. I tested with a guitar, music from smatphone and a sine wave generator.

Electrolytic capacitors (the 100nF in my case) have a polarity, the 33pF and 10 nF caps are ceramic and don't have a polarity. In the datasheet the polarity of the capacitors is given in the schematic, so I think its ok to use electrolytic ones.

NicoHood, 2017-07-03:
You should try another chip or different capacitors then.

pubcoder, 2017-07-04:
I noticed that there are two data sheets of MSGEQ7: older from 2004 and newer from 2011.
The newer uses only 33pF and 0.1u caps and no 10pF. Could that be significant?
I always constructed based on newer schematics and it always worked, well except some broken chips which lacked some frequences or had some frequencies always high.
Always order in 5 or more batches because they are a hit or miss.

Triscus, 2017-07-04:
I tried it first with the 0.1µF caps because I had no other caps at hand, but it's the same with the 10pF caps. I also wrote MSI, they stated it's no problem to use electrolytic caps as long as they are in the correct direction. 

Thank you fot your advice, I think I will order new chips and try again. If you let the issue open I can write down my outcome :)

Triscus, 2017-09-12:
The chips were ok, I used a wrong cap, 33nF instead of 33pF (Doh!) Because of that the frequency of the output was altered. So I actually read one band at a time instead of all seven.

NicoHood, 2017-09-12:
Thanks for the feedback and happy coding! :)

closed 2017-09-12 by NicoHood