NicoHood/MSGEQ7 #9: Compile error using Visual Micro (includes multiple libraries by you)

Ender06, 2017-08-03:
I'm trying to switch over my coding environment to Visual Studio/Visual Micro and have run into two problems, it seems to focus around your libraries.   I am able to see and install (most) of your libraries via the normal click/install route on the Visual Micro explorer.  But for your MSGEQ7 library it gives this error:  

Unable to install the MSGEQ7 library: 
As you can see, here is the error, also note, grey means I've successfully downloaded and installed that library:
[image]

Here is the screen showing that your libraries are installed:
[image]

However, here Visual Micro is not registering your libraries as actually "installed" (none of your libraries are in green):
[image]

I will try to contact Visual Micro about this, but I'm wondering if maybe something in your libraries are not configured right?  It's odd that only your libraries are doing this...

Ender06, 2017-08-03:
Somewhat of an update, I stumbled across a workaround, for some reason adding the library within the project folder, and using intellisense to add the include command worked.... 

See:
[image]
[image]

It looks like using intellisense allows for the compiler to work, (it looks like you must add the folder(s) within the library folder to get it to recognize the files.  So instead of "#include MSGEQ7.h" you must use "#include "MSGEQ7\src\MSGEQ7.h"".

I will contact Visual Micro regarding this.

NicoHood, 2017-08-03:
I think this is expected. The /src folder is the new layout for libraries in the arduino IDE. All of my libraries wont work, because all of them comply to the new format, and other devs did not switch yet.

The Arduino IDE searches for a /src folder and uses this as include path. If no /src is available, it uses the libraries root.

The problem I see here is, that adding all those files as include is dangerous, as if you have "util.h" this might conflict with other libs. But that is a problem of the arduino ide itself. But that is just a general problem.

closed 2017-09-12 by NicoHood