NicoHood/MSGEQ7 #12: Cannot get bands to separate to individual LEDS

chuco61, 2018-10-13:
Hi,

Im attempting to get the 4 bands to spread over individual LEDs. The code alone works very well with my msgeq7, now i'd like to have the mids every 3rd LED, the highs every 4th LED etc. Any help is appreciated, sorry I dont know how to contact you any other way with the provided methods you've provided.

chuco61, 2018-10-13:
``/*
  Copyright (c) 2014-2016 NicoHood
  See the readme for credit to other people.

  MSGEQ7 FastLED example
  Output via Led strip and FastLED library

  Reads MSGEQ7 IC with 7 different frequencies from range 0-255
  63Hz, 160Hz, 400Hz, 1kHz, 2.5kHz, 6.25KHz, 16kHz
*/

// FastLED
#include "FastLED.h"

#define LED_PIN    7 // DATA_PIN, or DATA_PIN, CLOCK_PIN
#define COLOR_ORDER GRB
#define CHIPSET     WS2811 // WS2811 LPD8806
#define NUM_LEDS    30

#define BRIGHTNESS  15  // reduce power consumption
#define LED_DITHER  0  // try 0 to disable flickering
#define CORRECTION  TypicalLEDStrip

CRGB leds[NUM_LEDS]; // Define the array of leds

// MSGEQ7
#include "MSGEQ7.h"
#define pinAnalogLeft A0
#define pinReset 3
#define pinStrobe 2
#define MSGEQ7_INTERVAL ReadsPerSecond(50)
#define MSGEQ7_SMOOTH true

CMSGEQ7<MSGEQ7_SMOOTH, pinReset, pinStrobe, pinAnalogLeft> MSGEQ7;

void setup() {
  // FastLED setup
  FastLED.addLeds<CHIPSET, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection(CORRECTION);
  FastLED.setBrightness( BRIGHTNESS );
  FastLED.setDither(LED_DITHER);
  FastLED.show(); // needed to reset leds to zero

  // This will set the IC ready for reading
  MSGEQ7.begin();
}

void loop() {
   
  bool newReading = MSGEQ7.read(MSGEQ7_INTERVAL); // Analyze without delay
 
  if (newReading) {  // Led strip output
    uint8_t val = MSGEQ7.get(MSGEQ7_BASS); // visualize the average bass of both channels
    val = mapNoise(val); // Reduce noise

  // Visualize leds to the beat
    CRGB color = CRGB::Blue;
    color.nscale8_video(val);
    fill_solid(leds, NUM_LEDS, color);
    // Update Leds
    FastLED.show();
  }

  if (newReading) {  // Led strip output
    uint8_t val = MSGEQ7.get(MSGEQ7_MID); // visualize the average bass of both channels
    val = mapNoise(val); // Reduce noise
  // Visualize leds to the beat
    CRGB color = CRGB::Red;
    color.nscale8_video(val);
    fill_solid(leds, NUM_LEDS, color);
    // Update Leds
    FastLED.show();
  }

  if (newReading) {  // Led strip output
    uint8_t val = MSGEQ7.get(MSGEQ7_HIGH); // visualize the average bass of both channels
    val = mapNoise(val); // Reduce noise
  // Visualize leds to the beat
    CRGB color = CRGB::White;
    color.nscale8_video(val);
    fill_solid(leds, NUM_LEDS, color);
    // Update Leds
    FastLED.show();
  }

    if (newReading) {  // Led strip output
    uint8_t val = MSGEQ7.get(MSGEQ7_LOW); // visualize the average bass of both channels
    val = mapNoise(val); // Reduce noise
  // Visualize leds to the beat
    CRGB color = CRGB::Green;
    color.nscale8_video(val);
    fill_solid(leds, NUM_LEDS, color);
    // Update Leds
    FastLED.show();
  }

 }

NicoHood, 2018-10-14:
You better ask such a question in the arduino forums or chat, as its a simple programming question, nothing specific to this library.

closed 2018-10-14 by NicoHood