NicoHood/MSGEQ7 #13: MSGEQ7 give wrong value

Cyba3rr, 2018-10-16:
Hey there,
the lib is doing very well. But if i start an Accsesspoint the int is still by 255 and no changes. If i stop the Webserver its good to go. Hope Some one helps or the bug will be fixed. I´m using an ESP32 DEV KIT 1.
Thanks a lot.

NicoHood, 2018-10-19:
I never tried the ESP with this library. Also the voltage levels must kill the chip, dont they? You must use an analog pin and If I remember correct, its 1.8v?

Cyba3rr, 2018-10-20:
Tanks for you're comment. Did you test the library with an access point? So when you click on a button on html, the arduino should change the colours while bass changed?

NicoHood, 2018-10-20:
No, because I never used an ESP.

Cyba3rr, 2018-10-22:
I've found the issue. The esp shutdown all gpios under 32 if WiFi get enabled.

closed 2018-10-22 by Cyba3rr