NicoHood/MSGEQ7 #19: Library broken with ESP8266 Core 2.7.0+

dlovinger, 2021-02-04:
This MSGEQ7 library appears to be broken on ESP8266 Core v2.7.0+ ([https://github.com/esp8266/Arduino#installing-with-boards-manager](https://github.com/esp8266/Arduino#installing-with-boards-manager)). 

A simple read using MSGEQ7.get(i) returns extremely large, periodically spiking values. 

Simply downgrading the ESP8266 version to 2.6.3 immediately solves the issues. However there are a number of network improvements on 2.7.0 that is essential to many users.

NicoHood, 2021-02-04:
Can you report this at their upstream or investigate the bug?

dlovinger, 2021-02-05:
I had a bit more time to investigate, and discovered that the map() function was the cause of the issue, which had changed in the ESP 2.7.x builds. The core MSGEQ7 library actually works fine it turns out, but I had some rather specific implementations using map(). I defined my own map() function to work around this. We can close this issue!

closed 2021-02-05 by NicoHood