NicoHood/MSGEQ7 #20: is this library commpatilbe with the esp32?

sebasdt, 2021-03-07:
it seems like a great library some filters and stuff like that.

NicoHood, 2021-03-08:
There were patches for the ESP:
https://github.com/NicoHood/MSGEQ7/pull/16

I think if analogRead is supported, it will work. You can report, if it works, so we can clarify the readme.

sebasdt, 2021-03-08:
thanks for your reply,  are those patches applied to the current version?

or do i need to add them? if so it's a bit unclear what to change, remove or leave at it is.

NicoHood, 2021-03-08:
Patches are already applied. What exactly are you missing? I am not sure what you mean.

sebasdt, 2021-03-08:
what I mean by this is I'm able to read from the chip no problem.

as shown by this screenshot, I only can get a maximum peak of  255.  do i need to change some variables?
[image]

NicoHood, 2021-03-08:
Oh wait, the PR was not merged, so no esp fixes applied yet...

However what is the problem with 8bit values? A higher resolution does not make sense in most cases. But I guess there could be still problems with the ESP. You might want to have a look at that closed PR

ErrorErrorError, 2021-03-08:
ESP32 works, just that you have to change the analog resolution to 8bit values using `analogResolution()`.

sebasdt, 2021-03-08:
normally I don't have an issue with lower resolution, it means fewer data to handle. in the case i use it i need the higher resolution.
if there's a simple setting to change that would be way easier. 
besides that, this library looks great has some useful features. credits to you!

sebasdt, 2021-03-08:
> ESP32 works, just that you have to change the analog resolution to 8bit values using `analogResolution()`.

oh i just saw this when i posted my post. thanks gona take a look at that.

ErrorErrorError, 2021-03-08:
We could try implementing macros that allow for certain boards, ESP32 Zero, Due, MKR family and Nano 33, that can have different resolution but well i'm not sure if that's the best way or just map it to 10 bit.